schollz/find #158: Docker saving data

pero86, 2017-04-04:
Hey guys,
I'm running FIND via Docker, and it works, but it does not save any changes. every time I restart my computer and run FIND via Docker again, all data is lost.

I'm using command listed in documentation:
`docker run -it -p 18003:8003 -p 11883:1883 -v /path/to/host/data/folder:/data schollz/find`
of course I'm replacing path to path on my local machine. 

No files are visible in folder specified in command above.
I'm not really familiar with docker but trying to solve this for a few days but without any success.

schollz, 2017-04-12:
Can you confirm that going to localhost:18003 shows that FIND is running on the Docker image?

mhoaglund, 2017-08-10:
Can confirm this.

When starting a fresh EC2 instance and following the Docker setup instructions and using the following run command from the /find directory:
docker run -it -p 18003:8003 -p 11883:1883 -v /data:/data finddocker

I am able to access FIND at port 18003 normally and everything works great! However, tellingly, the Export Data function isn't working in the portal (redirects to login) and no new files are appearing in that data directory or any other data directory on the vm.

schollz, 2017-08-11:
Yeah, I'm replicating this problem now. It seems like the `-data /data` is not being passed when Docker starts the `supervisord` process. I really don't know why this is, I will keep looking into it. 

The only thing I've noticed is that it maybe has something to do with `supervisord`...as you can run the bash of Docker and manually input the find command and it will save correctly.

schollz, 2017-08-11:
Yeah, it seems to be something with supervisord. The logfile seems to say this:

```
(version  build devdevde) is up and running on http://mosquitto`
```
which means it not understanding the flags passed to supervisor. I have no idea why. I've tried putting everything in a script and having supervisor run that, but that also doesn't work.

Here's a better workaround. Use the following to run:

```
docker run -it -p 18003:8003 -p 11883:1883 -v /path/to/host/data/folder:/root/find/data schollz/find
```

Because the flag `-data` is not being recognized from within docker, it will save to the default directory so you can just point your volume to that. Sorry about that.

mhoaglund, 2017-08-11:
Was just now able to corroborate your workaround. Thank you so much, Zack!

Maybe this level of specificity is unnecessary here, but this is the exact run command I went with. This works, assuming the user installed FIND in their home directory.

`docker run -it -p 18003:8003 -p 11883:1883 -v ~/find/data:/root/find/data finddocker`

closed 2017-10-01 by schollz