schollz/find #159: server setup errors

ksbickmore, 2017-04-12:
When I try to setup a findserver using mqtt I use the following command and get the listed warning.

`./findserver -mqtt 127.0.0.1:1883 -mqttadmin admin -mqttadminpass admin -mosquitto `pgrep mosquitto` -p :8003 192.168.0.1:8003
/home/user1/workspace/find/data
WARN : 2017/04/12 09:52:09 mqtt.go:164: exit status 1
DEBUG: 2017/04/12 09:52:09 mqtt.go:50: Finished setup`

After registering my group and when trying to use findclient I get another error related to my server.
I issue the command:
`./findclient -s 192.168.0.1:8003 -g mygroup -e -l myroom`

After the client runs iw the first time I receive an error:
`10:43:22.138 main  -  CRIT 00b Post 192.168.0.1:8003/learn: unsupported protocol scheme ""`

This happens when I use the docker image as well.

schollz, 2017-04-12:
Try adding "http://":

```
./findclient -s http://192.168.0.1:8003 -g mygroup -e -l myroom
```

ksbickmore, 2017-04-13:
yes that solved it

closed 2017-04-13 by ksbickmore