schollz/find #163: How does the find server guess the location

kmic25, 2017-04-21:
Hi,

I am trying to understand how the find-server guesses the location of a device.

My understanding so far:
- rssi is the signal strength of the device
- whilst in learning mode, the find-server receives multiple rssi values and the location name
- when tracking, the find-server receives an rssi and tries to match it to one of the locations

I am currently using find-lf and not the mobile app to send the rssi values. I have 3 Raspberry Pis in monitor mode which are all sending rssi values to the find server. In this case, I would expect signal strength to vary depending on which Raspberry Pi received the signal (i.e. if I'm exactly next to Pi 1 but 10m away from Pi 2, the rssi received by Pi 1 should be different than that of Pi 2). Two issues with this, the rssi value is actually not so different - there are many overlaps, and the find server will never receive anything that indicates whether the value is the signal strenght from Pi 1 or from Pi 2.

My main question is, how will the Find server be able to correctly guess the location if it never knows what the signal strength indicates (i.e. signal strength from which pi).

Any help is greatly appreciated. Thanks

schollz, 2017-04-22:
This sounds like an issue with find-lf. Have you tried resetting the Raspberry Pis? I've seen something similar before in some of my Raspberry Pis, where they stop working after a few days (i.e. not collecting unique signals anymore), but a simple reset seemed to fix it.

Also, are your two Pis in different rooms? Walls attenuate signals a lot, so expect to see bigger differences between rooms than within rooms (unless the room is huge).

kmic25, 2017-04-22:
Thanks for your prompt reply.

I have reset the Pis, but nothing seems to change. The 3 which are monitoring are set up in different corners of a fairly large open-plan office. I've tried using the app in the same office and results are quite accurate. It's with find-lf that I'm having issues.

However from what I understood the 'Fingerprint' struct does not include the IP address or some identifier to distinguish values sent from different Pis. So when the find-lf server sends the values to the Find server there doesn't seem to be any distinction.

schollz, 2017-04-22:
Oh! Did you change the hostname on your Pis? The hostname is used as the unique identifier which is compiled by the lf server.

kmic25, 2017-04-22:
Thanks a lot! No I didn't change that. Will try it out and get back to you with any progress.

schollz, 2017-05-17:
@kmic25 did you have any luck with this?

kmic25, 2017-05-17:
Hi, yes sorry for not getting back to you. I've updated the hostname and results have improved. A couple of days ago I've also set the Find server to use SVMs and RF which (especially using RF) has improved the results drastically.

Currently my main concern is with the Find-lf cluster since sometimes it stops sending any signals to the Find-lf server even though when checking the status it says they're all scanning and the find-lf server is still running.

schollz, 2017-05-17:
Yeah, I've run into this. I think it is a Raspberry Pi issue. My workaround was to have a cronjob restart the raspberry pi after a few hours and then restart sending (its only a few minute interruption). This seems to put them back on the regular scanning

kmic25, 2017-05-17:
Noted. Thanks!

closed 2017-05-17 by kmic25