schollz/find #167: Missing authentication support for existing MQTT server

MrMep, 2017-04-27:
If I understood well, when told to use a FIND created MQTT server configuration, the server requires user and password authentication and uses them.
But when using an existing MQTT server, user and passwords are ignored (`mqtt.go` line 33).

thanks,
gl

schollz, 2017-05-17:
Sorry, that was my mistake. It should be fixed now

MrMep, 2017-05-19:
I'm sorry, but I don't find any related changes in the source, what am I missing?
Perhaps I had to explain myself better. Here's what I did in my installation. I changed line 33 of mqtt.go from:
`                opts.AddBroker(server).SetClientID(RandStringBytesMaskImprSrc(5)).SetCleanSession(true)
`
to:
`opts.AddBroker(server).SetClientID(RandStringBytesMaskImprSrc(5)).SetCleanSession(true).SetUsername(RuntimeArgs.MqttAdmin).SetPassword(RuntimeArgs.MqttAdminPassword)`

so now I can launch find with:
`./find -mqtt localhost:1883 -mqttadmin find -mqttadminpass <secret> &`

This way I can connect to my existing mosquitto server, authenticating with username (`find` in may case) and password.
Obviously this isn't a solution, just a workaround. You might want to add two parameters `mqttuser` and `mqttuserpass` that are passed to line 33 of mqtt after proper check.
I'm sorry, I would do it myself but I don't know enough neither the project nor go, and you might want to find another solution to this problem.
thanks!

schollz, 2017-05-19:
So the whole MQTT thing is a bit of a workaround. The problem was that my server needed to start with the configuration file used by FIND. That configuration file specifies the password file and configuration.

The MQTT on the public server is a little complicated because it allows users to register themselves on the MQTT server. Its pretty hard to do that I've found, so my workaround was to give FIND admin access to MQTT and allow FIND to hot-reload the configuration file.

I'd love to have a better solution to all this, but its tricky because personal users don't really care about having to support a bunch of random people using their MQTT server, whereas I still do :|

MrMep, 2017-05-19:
I understand.
But in my personal installation I gave the mosquitto's `find` user just the privileges to read/write anything under my FIND group topic, something like:
`user find
topic readwrite myfindgroup/#`

That is working well so I think that, if you'd just add support for two additional parameters (mqttuser mqttpass), to be passed optionally only in case of an existing mosquitto service, you would solve a simple but, I guess, common problem:
- the way FIND is now, it requires an existing mosquitto allowing for anonymous access.
I think that is rare. I  think the most common situation is an existing setup already containing a mosquitto service that is very likely password protected.

I think you could add the management of the two parameters `mqttuser` and `mqtttuserpass` to `server.go` and then, in `mqtt.go`, line 33, if both parameter are not empty, add them to the options, otherwise not.

What do you think?

aherbjornsen, 2017-05-24:
Having spent a some time struggling with findserver, I can confirm that is an issue for me as well. Patching mqtt.go as outlined above works, but having options for username and password would make much more sense.

schollz, 2017-05-24:
@aherbjornsen @MrMep Okay, that sounds good. I'll give it a try myself too and then issue an update!

fab33, 2017-12-30:
Same problem here. Any news ?