schollz/find #170: Slow SVM causing HTTP timeouts

infernix, 2017-07-23:
Hi,

When using room+ to learn, it's getting timeouts and this is caused by svm-train:

```
Jul 23 15:19:20 spirit find[11003]: DEBUG: 2017/07/23 15:19:20 fingerprint.go:166: Inserted fingerprint containing 9 APs for gerben (b43) at office
Jul 23 15:19:20 spirit find[11003]: [GIN] 2017/07/23 - 15:19:20 | 200 |   54.789174ms |    192.168.0.15 |  POST     /learn
Jul 23 15:19:21 spirit find[11003]: DEBUG: 2017/07/23 15:19:21 mqtt.go:182: Got valid MQTT request for group b43, user gerben
Jul 23 15:19:21 spirit find[11003]: DEBUG: 2017/07/23 15:19:21 fingerprint.go:211: Was learning, calculating priors
Jul 23 15:19:23 spirit find[11003]: DEBUG: 2017/07/23 15:19:23 svm.go:232: svm-train -s 0 -t 0 -b 1 CbkabyZIWtZlzGHC.full data/b43.model
Jul 23 15:19:47 spirit find[11003]: DEBUG: 2017/07/23 15:19:47 svm.go:239: svm-train -s 0 -t 0 -b 1 IzmpHohMudnSqSga.learning IzmpHohMudnSqSga.learning.model
Jul 23 15:19:53 spirit find[11003]: DEBUG: 2017/07/23 15:19:53 svm.go:246: svm-predict -b 1 tGDMfntwgQtFhRli.testing IzmpHohMudnSqSga.learning.model yCYyuvQFoQWMnAyz.out
Jul 23 15:19:53 spirit find[11003]: DEBUG: 2017/07/23 15:19:53 svm.go:251: b43 SVM: Accuracy = 92.9314% (1341/1443) (classification)
Jul 23 15:19:53 spirit find[11003]: DEBUG: 2017/07/23 15:19:53 utils.go:109: TIMEING  took  30.180413444s
Jul 23 15:19:53 spirit find[11003]: DEBUG: 2017/07/23 15:19:53 rf.go:38: Writing b43.rf.json
Jul 23 15:19:53 spirit find[11003]: DEBUG: 2017/07/23 15:19:53 rf.go:67: RF classification success for 'b43' is 0.97
```

svm-train is taking a very long time and it's doing so at max CPU on a Xeon(R) E5-1620 v2 @ 3.70GHz. This in turn causes the HTTP request to time out.

Is it expected that svm-train takes 30 seconds to complete? Can its performance be increased and/or what causes it to be relatively slow? And if inevitable, can this not be done on a background thread?

schollz, 2017-07-24:
Can you let me know which version you are using? If building from source, the latest commit hash, if downloaded a binary the version of the binary.

infernix, 2017-07-24:
Master as of yesterday, on Debian 8. I am going to upgrade to 9 to see if it makes a difference (for svm-tools)

schollz, 2017-07-24:
There are some parameters to tune svm, but I believe the master branch should be the fastest. Actually, if its really slowing you down I would just not use it as RF and NB are both much faster and more accurate. To stop using, you can just uninstall the svm-* programs and FIND will detect that they aren't there and not use them.

infernix, 2017-07-26:
Ok, ive deleted the db, replaced my one AP with two new ones, and there are usually about a dozen aps in range per fingerprint.  But it seems as if the accuracy without svm went down. Is that possible?

schollz, 2017-07-26:
Accuracy may have changed when if it changed the parameters used. You can tune the parameters yourself also, see the docs for info.

The best strategy depends on the size of the room. For smaller rooms I think its sufficient to put the phone in the center.

closed 2017-07-26 by infernix