schollz/find #171: Adding new user to an existing group.

OlegSl, 2017-07-25:
Hello.
Now I'm working on Android mobile client for FIND. 
As I understand the only way to add new user to an existing group is to execute "/learn" request.
Could you add new request to API? Something like

**POST** /username
params:

group | Defines the unique group ID | Required
user    | Specifies a user                      | Required

I think this is quite useful because in this case user can just specify the name of "device" to start using previously created WiFi fingerprints map.

P.S. Thank for you work! The project you started is awesome!

OlegSl, 2017-07-26:
Hello again :-)
The main idea of my request is to have a possibility to track user when he send "/track" request to the FIND server. Description of use case is
1) admin creates "map" by "/learn" request;
2) admin installs mobile client on user's devices;
3) mobile client send "track" request to FIND server;
4) admin can get last known location for each user by "/location" request.

But I find out that FIND server updates user location only after "/learn" request.

So I will try to add additional logic to existing server project or use different service to store results of client "/track" request.

schollz, 2017-07-26:
Short answer: new users will automatically show up in the `/location` list, after about 30 seconds.

Long answer: Everytime you `GET /location` you run [`getUserLocations()`](https://github.com/schollz/find/blob/master/api.go#L397-L444). This code gets the location of all the users using [`getUsers()`](https://github.com/schollz/find/blob/master/db.go#L51-L82). This function defaults to the [users cache](https://github.com/schollz/find/blob/master/cache.go#L91-L97), if it exists it uses that. When you have a new user that is being tracked, it will start showing that user once the cache resets. Currently [that is set to about 30 seconds](https://github.com/schollz/find/blob/master/cache.go#L40-L45). I suggest setting this to much shorter if you need to update quickly. The reason the cache exists is because I am getting requests on the public server at a very high rate, and I want to avoid looking up the user positions for every user (especially if they are deep in the DB) at every request.

schollz, 2017-07-27:
There is a problem, but I think its not with adding a new user, as I can't replicate your problem with other groups.

There seems to be a problem with your learned locations: https://ml.internalpositioning.com/dashboard/PLUMPPIKE. They aren't there anymore, did you delete them?

OlegSl, 2017-07-27:
Oh....
It seems that for some requests groupName is case sensitive and for other isn't.
Just compare 
https://ml.internalpositioning.com/dashboard/PLUMPPIKE
with
https://ml.internalpositioning.com/dashboard/plumppike.

I'm not sure is that "bug or feature".

Thank you for help anyway. :-)

OlegSl, 2017-07-27:
[Here](https://github.com/schollz/find/blob/master/api.go#L385) group name is always in lowercase.
[Here](https://github.com/schollz/find/blob/master/fingerprint.go#L111) isn't.

OlegSl, 2017-07-27:
[Here](https://github.com/schollz/find/blob/master/api.go#L385) group name is always in lowercase.
[Here](https://github.com/schollz/find/blob/master/fingerprint.go#L111) isn't.

Unfortunately i'm not good at GoLang so I can't create a pull request to fix that problem.

closed 2017-07-31 by OlegSl