schollz/find #172: Differentiate between 2.4 and 5ghz?

infernix, 2017-07-27:
So I think I have found why i am getting much worse results with two new Ubiquiti APS vs a single Asus AP:

[image]

The same mac is sending out signal on 2.4 and 5ghz with quite different RSSI values.

As I understand it, a fingerprint is the MAC  + RSSI, so in this case there would be two fingerprints per AP with quite different RSSI values for a single room, correct? If so, can something be done? E.g. could the fingerprint be extended with a single byte to indicate the band (seeing as 60ghz is starting to appear as well)?

Or is there another way to account for this? I'd prefer not to filter them out as they are the main APs in the house and I wouldn't want to rely only on neighbouring APs.

infernix, 2017-07-27:
Thinking this through, I realize it may be hard to add an additional byte to the fingerprint, so perhaps instead the first byte can be manupulated - say add 1 for 2.4Ghz bands, 2 for 5Ghz bands, and so forth?

infernix, 2017-07-27:
Never mind, they are different. I need to drink my coffee, sorry for the noise

closed 2017-07-27 by infernix

schollz, 2017-07-27:
haha no problem, please let me know if you have more questsion!