schollz/find #173: Filtering out randomized MAC addresses

victorhooi, 2017-08-20:
Is there any way we could filter out randomized MAC addresses? (e.g. from iOS 8+, Android 6.0+, Windows 10 etc.).

I believe on iOS, they set the U/L bit to 1, so that should be reasonably easy to filter out?

https://en.wikipedia.org/wiki/MAC_address#Universal_vs._local

schollz, 2017-11-13:
Hi @victorhooi , I still haven't figure out a way to do this. I was playing around with Scapy but had no luck. Have you found anything?

victorhooi, 2017-11-16:
Can we not just check for the U/L bit and add a filter on the web interface to show/hide MAC addresses with it set?

schollz, 2017-11-16:
That sounds like it would work, but I don't know how to do that!

I've been working on this to get raw packets: https://gist.github.com/schollz/ba151ba1fbb1bd433fd523ac9502947a

But I'm not sure how to extract the U/L bit