schollz/find #176: Not seeing all devices?

sudochownrwhoami, 2017-09-22:
I have a find server running locally on a machine running ubuntu.

I also have initialized (with the script in the docs for creating a cluster) two raspberry pi devices in different rooms, each has a tplink wifi adapter.

I copied the generated ssh key from the ubuntu machine to the 2 raspberry pi devices.

I then ran the `python3 cluster.py initialize` command, making sure to point to my ubuntu machine's lfaddress.

Finally I run `python3 cluster.py start` on the ubuntu machine and then `python3 cluster.py status` and it tells me both pis are scanning.

When I navigate in the browser to my local server's dashboard, I only see the user that has done learning via my Android app, but I don't see any other devices at all. 

Using the cluster, shouldn't I be able to see all wifi enabled devices? I have several wifi enabled devices in my home. What might I be doing wrong? Did I miss anything?

schollz, 2017-09-22:
Did you goto the lf server and enter your group name and change it to "track"?

sudochownrwhoami, 2017-09-22:
You mean logging in via GROUPNAME then clicking orange/yellow "Track" button in the Dashboard under "Users"? Yes, but it only shows the Android device I set to tracking, it doesn't show any other devices.

sudochownrwhoami, 2017-09-22:
Okay so I realize you were talking about running the find-lf server itself. Would my cluster config point to the find-lf server or the find server? How do I make the find server aware of the find-lf server? I must have missed something somewhere, thanks for your help, schollz

sudochownrwhoami, 2017-09-22:
so it looks like the raspberry pi devices aren't sending over any data, when I run scan.py manually, it says `2017-09-22 20:47:21,932 - process_scan:91 - DEBUG - Processed 0 lines, found 0 fingerprints in 0 relevant lines
2017-09-22 20:47:21,933 - process_scan:98 - DEBUG - {'node': 'raspberrypi', 'timestamp': 1506113241, 'signals': []}`
any ideas?

sudochownrwhoami, 2017-09-22:
So, after diving into it even more, it looks like tshark is saying that my tplink wifi adapter that I bought (the one mentioned in your docs) does not support monitor mode. however, when I run "iw list" it indicates that it DOES support monitor mode. 

what's going on here?

schollz, 2017-09-22:
I'm not sure why `iwlist` lists adapters that support monitor mode, while they actually don't (my onboard laptop chip does the same).

I'm sorry your tplink doesn't work. You are the second person to note that the current version of the tplink doesn't work. Another user told me that tplink had changed its chipset in the past year, but I hadn't verified it. I will update the docs now to prevent this in the future. I'm not sure which do now. To find the tplink originally I used this compendium: http://elinux.org/RPi_USB_Wi-Fi_Adapters and find ones that support "Ad Hoc" mode.

There is also progress with manipulating the onboard pi3 chip: https://github.com/seemoo-lab/nexmon. I haven't been able to check that out yet though.

sudochownrwhoami, 2017-09-26:
I actually ended up getting a Panda UBS adapter (model is PAU04) and that works nicely. One thing I'm trying to figure out is why my users look like this:
`sent as /track
{
 "group": "bison",
 "username": "0011323505c5",
 "location": "unknown",
 "timestamp": 1506450382915613844,
 "wifi-fingerprint": [
  {
   "mac": "raspberrypi",
   "rssi": -55
  }
 ]
}`
Why is the mac address "raspberrypi"? And when I change the username for a device, why doesn't it stay there when it gets updated?

Thanks for your help!

schollz, 2017-09-28:
The mac address is the hostname. The reason I did this is that you can match up easily when searching for LAN devices on the network. You can, and should, change the hostname if you have more than one Pi.  

That's great that the Panda USB works! Would you like to make a PR to the FIND-LF to update the Readme with that info?

sudochownrwhoami, 2017-09-29:
just made the PR

schollz, 2017-10-03:
Thanks, should I close this?