schollz/find #177: fingerprint binary won’t run on macos high sierra

diplix, 2017-09-26:
```
ix$ /Users/ix/bin/fingerprint/fingerprint -s http://192.168.xx.xx:8003 -u ix -g ix -c 1
failed MSpanList_Insert 0x72b000 0xd8df01c9e37 0x0 0x0
fatal error: MSpanList_Insert

runtime stack:
runtime.throw(0x4bfbd0, 0x10)
	/usr/local/go/src/runtime/panic.go:530 +0x90 fp=0x7ffeefbff100 sp=0x7ffeefbff0e8
runtime.(*mSpanList).insert(0x698a48, 0x72b000)
	/usr/local/go/src/runtime/mheap.go:933 +0x293 fp=0x7ffeefbff130 sp=0x7ffeefbff100
runtime.(*mheap).freeSpanLocked(0x698240, 0x72b000, 0x100, 0x0)
	/usr/local/go/src/runtime/mheap.go:809 +0x4be fp=0x7ffeefbff198 sp=0x7ffeefbff130
runtime.(*mheap).grow(0x698240, 0x8, 0x0)
	/usr/local/go/src/runtime/mheap.go:675 +0x2a0 fp=0x7ffeefbff1f0 sp=0x7ffeefbff198
runtime.(*mheap).allocSpanLocked(0x698240, 0x1, 0x0)
	/usr/local/go/src/runtime/mheap.go:553 +0x4e3 fp=0x7ffeefbff248 sp=0x7ffeefbff1f0
runtime.(*mheap).alloc_m(0x698240, 0x1, 0x15, 0x0)
	/usr/local/go/src/runtime/mheap.go:437 +0x119 fp=0x7ffeefbff278 sp=0x7ffeefbff248
runtime.(*mheap).alloc.func1()
	/usr/local/go/src/runtime/mheap.go:502 +0x41 fp=0x7ffeefbff2a8 sp=0x7ffeefbff278
runtime.systemstack(0x7ffeefbff2c8)
	/usr/local/go/src/runtime/asm_amd64.s:307 +0xab fp=0x7ffeefbff2b0 sp=0x7ffeefbff2a8
runtime.(*mheap).alloc(0x698240, 0x1, 0x10000000015, 0x156df)
	/usr/local/go/src/runtime/mheap.go:503 +0x63 fp=0x7ffeefbff2f8 sp=0x7ffeefbff2b0
runtime.(*mcentral).grow(0x699e40, 0x0)
	/usr/local/go/src/runtime/mcentral.go:209 +0x93 fp=0x7ffeefbff360 sp=0x7ffeefbff2f8
runtime.(*mcentral).cacheSpan(0x699e40, 0x692b98)
	/usr/local/go/src/runtime/mcentral.go:89 +0x47d fp=0x7ffeefbff3a0 sp=0x7ffeefbff360
runtime.(*mcache).refill(0x727000, 0x15, 0x7ffeefbff408)
	/usr/local/go/src/runtime/mcache.go:119 +0xcc fp=0x7ffeefbff3d8 sp=0x7ffeefbff3a0
runtime.mallocgc.func2()
	/usr/local/go/src/runtime/malloc.go:642 +0x2b fp=0x7ffeefbff3f8 sp=0x7ffeefbff3d8
runtime.systemstack(0x7ffeefbff498)
	/usr/local/go/src/runtime/asm_amd64.s:307 +0xab fp=0x7ffeefbff400 sp=0x7ffeefbff3f8
runtime.mallocgc(0x180, 0x46e780, 0x0, 0x800000000)
	/usr/local/go/src/runtime/malloc.go:643 +0x869 fp=0x7ffeefbff4d8 sp=0x7ffeefbff400
runtime.newobject(0x46e780, 0x693170)
	/usr/local/go/src/runtime/malloc.go:781 +0x42 fp=0x7ffeefbff500 sp=0x7ffeefbff4d8
runtime.malg(0x8000, 0x693520)
	/usr/local/go/src/runtime/proc.go:2634 +0x27 fp=0x7ffeefbff538 sp=0x7ffeefbff500
runtime.mpreinit(0x693ac0)
	/usr/local/go/src/runtime/os1_darwin.go:140 +0x1f fp=0x7ffeefbff550 sp=0x7ffeefbff538
runtime.mcommoninit(0x693ac0)
	/usr/local/go/src/runtime/proc.go:494 +0x105 fp=0x7ffeefbff598 sp=0x7ffeefbff550
runtime.schedinit()
	/usr/local/go/src/runtime/proc.go:434 +0x79 fp=0x7ffeefbff5e0 sp=0x7ffeefbff598
runtime.rt0_go(0x7ffeefbff610, 0x9, 0x7ffeefbff610, 0x0, 0x9, 0x7ffeefbff7d0, 0x7ffeefbff7f6, 0x7ffeefbff7f9, 0x7ffeefbff812, 0x7ffeefbff815, ...)
	/usr/local/go/src/runtime/asm_amd64.s:138 +0x132 fp=0x7ffeefbff5e8 sp=0x7ffeefbff5e0
```

schollz, 2017-09-26:
Thats strange. It looks like a Go issue. Did you download the binary or compile it yourself? 

I've updated the binary so you can try again by downloading here: https://github.com/schollz/find/releases/tag/v0.6client

diplix, 2017-09-26:
i downloaded the binary and the new one (0.6) works again. tnx!

closed 2017-09-26 by diplix