schollz/find #180: MQTT is not enabled on this server

SaminKainat, 2017-10-09:
while configuring FIND server, i have tested my mqtt server using this command 
mosquitto_pub -d -t hello/world -m
 This is successfully running.

but, when i write this command

 curl -X PUT 127.0.0.1:8003/mqtt?group=test, 
i get the following error.

{"message":"MQTT is not enabled on this server","success":false}

What could be the reason of this error? if anybody can help?

schollz, 2017-10-09:
Did you add the `-mqtt` flag when you started the FIND server? See [these instructions](https://www.internalpositioning.com/server/#mqtt-support).

SaminKainat, 2017-10-09:
I think not.
Can you please tell me how to add mqtt flag?

schollz, 2017-10-09:
See the link above. You can set the flag when you run the server

SaminKainat, 2017-10-10:
Thankyou for your response. 
Resolved my error through this command
./find -mqtt MQTTSERVER:1883 :FINDPORT

closed 2017-10-10 by schollz