schollz/find #182: Get x,y,z Cartesian coordinates?

NicksonYap, 2017-10-23:
Hi,

I realise the final output is discrete, meaning it only states the room the device is in.
But, is it possible to obtain XYZ coordinates based on the RF (random forest) result?

[image]

Say, I have a floor plan:
[image]

And the result is room 3, but would it probably give an output saying:
`Room 3: 70%, Room 1: 10% , Room 2: 10%, Room 4: 10%`

Could we use the numbers above and somehow obtain X Y Z coordinates?
My concern is that if this is logically accurate, just want to hear some ideas.

Also, if anyone knows any references or repo that already provides X Y Z coordinates based on BSSID (mac) and RSSI, that would be nice :)

schollz, 2017-10-23:
I haven't really implemented coordinates yet, but there is a hack to do it: https://github.com/schollz/find-maraudersmap

NicksonYap, 2017-10-26:
@schollz was excited to try out but then realise the coordinates are faked with random number...
This isn't helpful at all.

Is my proposal above possible?
To do in a mathematical way.

Best Regards

schollz, 2017-10-26:
I'm not sure I understand. Do you want the coordinate of room 3? Or you want to determine the coordinate within room 3?

NicksonYap, 2017-10-26:
Sorry, "fake" is probably a wrong word to use.

I'd like to determine the coordinate within the rooms.

schollz, 2017-10-26:
That's okay. "Within" the rooms is probably not possible, or at least very very hard. The RSSI values certainly correlate with distance but the exactly function of distance can vary a lot depending on the strength of the detector and the surrounding environment.

NicksonYap, 2017-10-26:
I see, that makes sense since finding source codes for this itself seems pretty hard.

Btw, would it work if I learn the locations within the rooms?
Like, segment a room into 4 parts (or more)

To increase accuracy, I can place a number of small battery-powered device (ESP8266) to broadcast it's WiFi beacon in a single room.