schollz/find #184: Problem during local and docker installation

hellowtisch, 2017-10-27:
When running **sudo docker build -t finddocker .** this ends with the below error message. I do receive the same error when trying to install according to the local setup method...

```

Removing intermediate container 855f5513fa75
Step 22/33 : RUN git clone https://github.com/schollz/find.git
 ---> Running in a91d8f58bc9e
Cloning into 'find'...
 ---> ab81ed7c2d67
Removing intermediate container a91d8f58bc9e
Step 23/33 : WORKDIR "/root/find"
 ---> 311e3252cef5
Removing intermediate container e0de78c10153
Step 24/33 : RUN go build
 ---> Running in 51d7373aea37
api.go:21:2: cannot find package "github.com/boltdb/bolt" in any of:
	/usr/lib/go-1.6/src/github.com/boltdb/bolt (from $GOROOT)
	/usr/local/work/src/github.com/boltdb/bolt (from $GOPATH)
routes.go:20:2: cannot find package "github.com/gin-gonic/contrib/sessions" in any of:
	/usr/lib/go-1.6/src/github.com/gin-gonic/contrib/sessions (from $GOROOT)
	/usr/local/work/src/github.com/gin-gonic/contrib/sessions (from $GOPATH)
api.go:22:2: cannot find package "github.com/gin-gonic/gin" in any of:
	/usr/lib/go-1.6/src/github.com/gin-gonic/gin (from $GOROOT)
	/usr/local/work/src/github.com/gin-gonic/gin (from $GOPATH)
fingerprint_ffjson.go:11:2: cannot find package "github.com/pquerna/ffjson/fflib/v1" in any of:
	/usr/lib/go-1.6/src/github.com/pquerna/ffjson/fflib/v1 (from $GOROOT)
	/usr/local/work/src/github.com/pquerna/ffjson/fflib/v1 (from $GOPATH)
mqtt.go:23:2: cannot find package "github.com/schollz/org.eclipse.paho.mqtt.golang" in any of:
	/usr/lib/go-1.6/src/github.com/schollz/org.eclipse.paho.mqtt.golang (from $GOROOT)
	/usr/local/work/src/github.com/schollz/org.eclipse.paho.mqtt.golang (from $GOPATH)
The command '/bin/sh -c go build' returned a non-zero code: 1
```

schollz, 2017-10-27:
Hi @hellowtisch, found the issue. The issue was that I wasn't using the Gopath to build FIND in Docker. I needed to use the Gopath now, versus before, because now all the source is from vendored. It should be working now.

Also, I think you can also do `docker pull schollz/find`, which should be up-to-date.

schollz, 2017-10-27:
It should be fixed in 782ece7, but let me know if now.

hellowtisch, 2017-11-03:
unfortunately this still does not work, however the error message now is different: 

```
E: Failed to fetch http://security.ubuntu.com/ubuntu/pool/main/l/linux/linux-libc-dev_4.4.0-97.120_amd64.deb  404  Not Found [IP: 91.189.91.26 80]

E: Unable to fetch some archives, maybe run apt-get update or try with --fix-missing?
The command '/bin/sh -c apt-get install -y python3 python3-dev python3-pip' returned a non-zero code: 100

```

hellowtisch, 2017-11-03:
I was able to solve this by adding another `RUN apt-get update` prior to adding the Python stuff...

closed 2017-11-03 by hellowtisch

schollz, 2017-11-03:
@hellowtisch Great! Could you make a PR with that?