schollz/find #185: Scalability? Long wait for return from /location

KennethLowe, 2017-10-27:
Hi,

I've been evaluating FIND for use at a small company, and so far it's looked good. However now that I have 5 locations plotted and 4 devices to track my server is taking 40 seconds to return when I hit the /location end point. Every other endpoint seems fairly responsive. Also I am unable to filter the endpoint by user reliably. The majority of the time, if I specify a user I get back:

{
message: "You should insert fingerprints before tracking, see documentation",
success: false
}

But every once in a while I will get a speedy return of just the user's location that I asked for. It seems like something must be wrong here. But just to make sure, have you tested FIND with large datasets? I would imagine I am going to get to around 50-70 locations and 20-40 trackers, would it be reasonable for me to expect  ~1 second returns once I reach that volume of data? Do you have any suggestions for improving the performance I am seeing now?

Thank you for your time!
Kenny Lowe

schollz, 2017-10-27:
Are you using the public server?

Also about how many wifi access points are there? You can get these details
by clicking on on a location in the dashboard.

KennethLowe, 2017-10-27:
Wow, thank you for getting back to me so quickly!

I am hosting my own server, on a fairly beefy workstation with 40 cores and 128GB of RAM. Though I am running it dockerized and I'm not very familiar with docker and how it allocates my host systems resources. Hitting the status endpoint does display 40 cores though.

Depending on where in the building I am each fingerprint contains 20-35 access points.

I am wondering if this could be related to my companies network setup? Each base station is setup to broadcast 3 different networks. Each one seems to have a discrete MAC even though they are all coming from the same enterprise base station. When I click on a location I can see clearly on the graph that though there may be 35 base stations within range of the location, they are clustered in groups of 2-3, that have identical RSSI but differing MACs.

schollz, 2017-10-27:
@KennethLowe It could be the number of APs. The main ML algorithm uses Naive Bayes and it scales with NumberAccessPoints * NumberLocations. Still, I'm surprised that `/location` is slow though, as that route is pretty well cached and I've used FIND at Duke University with hundreds of APs at a single location with dozens of locations. Would you mind sending me your database file? In the `/data` directory you should find your `GROUPNAME.db`. You can send it to me at hypercube.platforms@gmail.com. I can take a look and see if there is something weird going on.