schollz/find #186: Problem with cluster.py initialize

DomKarb, 2017-11-03:
Hi,

I trying to setup find on 6x rpi zero w.

When I trying to initialize cluster, I've got this problem.

`pi@pi001:~ $ python3 cluster.py initialize
Enter Pi address (e.g. pi@192.168.1.2. Enter blank if no more): pi@192.168.0.133
Enter Pi notes (for you to remember): pi002
Which wlan to use (default: wlan1)?: wlan0
Enter Pi address (e.g. pi@192.168.1.2. Enter blank if no more):
Enter lf address (default: lf.internalpositioning.com):
Enter a group: Selectivv
Enter a scanning time (default 10 seconds):
copying ips`

I tried with few option on few machines.

Any ideas? 

Thanks in advance.

Best regards
Dominik

schollz, 2017-11-03:
What's the problem? There is no error in what you posted

DomKarb, 2017-11-03:
Sorry! I didn't paste whole code

`pi@pi001:~ $ python3 cluster.py initialize
Enter Pi address (e.g. pi@192.168.1.2. Enter blank if no more): pi@192.168.0.133
Enter Pi notes (for you to remember): pi002
Which wlan to use (default: wlan1)?: wlan0
Enter Pi address (e.g. pi@192.168.1.2. Enter blank if no more):
Enter lf address (default: lf.internalpositioning.com):
Enter a group: Selectivv
Enter a scanning time (default 10 seconds):
copying ips
Could not connect to {'wlan': 'wlan0', 'address': 'pi@192.168.0.133', 'notes': 'pi002'}`

Bassicly I saw information that I can't connect... dosn't metter which PI am I using. 

Shall I answer for all question and fill-in all PI's, and after 'connection error' should I used python3 cluster.py start and ignore the error?

DomKarb, 2017-11-04:
Ok, resolved. I ignore this communicate and moving forward ;)

closed 2017-11-04 by DomKarb