schollz/find #188: Segmentation fault: 11

robkamp, 2017-11-08:
I downloaded the findclient for Mac. After extracting the client and executing it fails with "Segmentation fault: 11".

I run macOS High Sierra Version 10.13.2.

schollz, 2017-11-08:
Which version of the FIND client? Can you run in debug mode and paste the output?

robkamp, 2017-11-08:
It is [version 0.5](https://github.com/schollz/find/releases/download/v0.5/findclient_0.5_osx.zip) as downloaded from the site. 

How do I start with debug. Every option, like --debug or -debug or --help, leads to the same segmentation fault. 

I downloaded a fresh copy extracted it. Same result.

schollz, 2017-11-08:
Hm I'm not sure. Can you install from source?

robkamp, 2017-11-09:
Compiled it and now it works.

chinghanyu, 2018-02-03:
Noted that there are two RF.py in the same folder. One in uppercase and one in lower case. This causes cloning and naming issues on Windows system. Please fix it.