schollz/find #189: Data representation (map, floor, etc)

Bohmaster, 2017-11-08:
Hi! First of all, this software is great! Thank you so much for sharing it.

I'm confused about one thing:

In the FAQ says:

> **Can I track myself on a map?**
> 
> Not yet. This is something we would like to implement and we are working on. There is more information about our longterm roadmap here.

In plugins:

> **Using FIND with a map**
> 
> The find-map is a simple setup for projecting the current locations using GPS coordinates. For more information, see find-map README.
> 

So, my question is: What findmap is for? Why there is no documentation or instructions inside README.md? Is it working?

If not, is there a way to represent the location in some kind of map (or floor) with an external plugin o third-party software (free or paid)?

Thank you in advance.

schollz, 2017-11-08:
Thanks. findmap is deprecated. I will update the repo. 

There is a way to represent the location, its a bit hacky: https://github.com/schollz/find-maraudersmap.

Bohmaster, 2017-11-09:
@schollz Great. I've tried to insall it locally but I get this: https://github.com/schollz/find-maraudersmap/issues/9