schollz/find #191: Minimum WiFi routers for normal accuracy

Bohmaster, 2017-11-09:
I've been using the APP with only 1 WiFi router and I'm getting some erratic behaviour. In some cases the location is totally erroneous and opposite to the place I'm in. 

So I suspect that I need a minimum of 3 WiFi router (I've been using this in my home. Not in wide areas) to get a better accuracy, right?

I've left my cellphone for 5 minutes doing the learning. How much is the minimum time to do the tracking?

Basically my question is which is the best way and context to get the most of this software. 

Thank you in advance.

Bohmaster, 2017-11-13:
@schollz Another question: Which RaspberriPi's are needed to use it as an access-point? I have two Pi Zero's (not Zero W) and one Pi 2 B.

Are, for example, the TPLink extender/access point a better choice? What do you recommend for large areas (100.000mts2)

schollz, 2017-11-13:
You can use any Pi for an access point, but you will need to get a WiFi card capable of it. There are some listed here: https://github.com/schollz/find-lf#requirements. The cheap ones work pretty well, and they have a great range.

How much time to do learning? I think 5 to 10 minutes is plenty. I would also add a learning session the following day to normalize any abnormalities that can occur from a one-time learning.

Sorry for the slow responses @Bohmaster , please feel free to message me if you have more questions or email me at zack.scholl@gmail.com

Bohmaster, 2017-11-13:
@schollz No problem Zack! Thank you for responding. I'll surely email you with specific questions. Great job my friend!