schollz/find #192: Wifi MAC white list

misisnik, 2017-11-10:
Is possible to define white list of Wifi APs which has to be included into the learning??
Because many of our APs in off the office are just some mobiles etc.. which are not alive all the time so we need to eliminate this MAC addresses.
Thanks Mike

schollz, 2017-11-13:
Yes, you can do that. Make a JSON of the mac to filter, like

```json
{
"ab:cd:ef":true,
"ab:cd:eg":true
}
```

and then run the FIND server with the `-filter whitelist.json`.

misisnik, 2017-11-15:
Thank you, but how can I know that is working  fine. When I have clicked on "Update" button in administration, it will generate me the group file (grid_4.rf.json), which contains all MAC address as before. Also the same in new group.

misisnik, 2017-11-24:
ok i put the -filter /tmp/whitelist.json nex to ..../find and it seems to work. But when i check .rf.json there are different mac address than address in whitelist but whi ? THX