schollz/find #193: About Learn

Bohmaster, 2017-11-13:
I'm having some questions about the Learn process:

1) Take for example a room (call it Area 1) of 4mts2. To do the learning do I need to stay quiet in the center of the room or move around?

2) If I'm in the center of Area 1, which is the radio of the learning process? I need to learn a area of 100000mts2 and I was thinking to divide it in multiple areas. My questión is how do I know how to divide it since I don't know which is the reach of the learning process (presumming the access points are well distributed and in the right amount in relation to the space)

schollz, 2017-11-13:
1. I always suggest moving around, if it is a big room. For not so big places, I think just do the learning in the middle.

2. I would start dividing into smaller areas and then you can look at the diagnostics (the distributions of the signals) and make a judgment of whether to combine them. To combine them you just rename one place to the other place.

Bohmaster, 2017-11-13:
Great. I will give it a try. 

Finally I will be using these access points

http://www.tp-link.com/us/products/details/cat-4908_EAP115.html

What do you think about it?

Cheers!

schollz, 2017-11-13:
Yes, those should work fine!