schollz/find #194: NO FINGERPRINTS FOUND TO INSERT, SEE API

tushe, 2017-11-25:
I tried to compile and install it on my phone and I got that error when I tried to send fingerprint...

schollz, 2017-11-25:
Can you verify that your WiFi is enabled? Also, are their any WiFi APs in
your vicinity?

tushe, 2017-11-25:
Hi @schollz I've WIFI enabled and I've a WiFi APs in my vicinity, with the play store version it works, with my compiled version it doesn't work :(

schollz, 2017-11-25:
The compiled app doesn't work? Which repo did you compile?

tushe, 2017-11-25:
This repo with cordova

tushe, 2017-11-25:
I added the plug-in and i compiled it, I installed the apk on my smartphone but it can't send fingerprints...

schollz, 2017-11-25:
Hm. There are a couple things, could you setup a local server and see if
it's connecting at all? I think most likely it's a Android permissions
issue. The Google Play app seems to be grandfathered into the old
permissions scheme, but they changed it on next Android versions where the
app needs to explicitly request permissions.

tushe, 2017-11-25:
Ok, I was setup a local server but the result was the same. 
If I'll have time I could develop a new version of this Cordova based app and i'll share it. 
Thank you for your help.

schollz, 2017-11-25:
You might check out https://github.com/uncleashi/find-client-android, it
works well too, and is more up to date.

tushe, 2017-11-26:
Thanks, I compiled that native app and tested it, but seem it has some issue about determining the position.

ishanjain28, 2017-11-27:
@schollz I am having this same problem on Find's Android Application. The server is https://ml.internalpositioning.com.

schollz, 2017-11-27:
> internalpositioning.com.

Did you do learning? Could you send me your group name?

tushe, 2017-11-27:
I solved my issue with compiled cordova app just adding this in AndroidManifest.xml:

<uses-permission android:name="android.permission.ACCESS_FINE_LOCATION" />

After installing go to Settings -> Applications -> Find -> Authorization  and enable Position.

ishanjain28, 2017-11-27:
@schollz I did learning/tracking both were giving same error. Then I tried running `findclient` from my PC and that worked just fine. 

My group name is "ishanjain28group"

tushe, 2017-11-27:
I solved the issue with my cordova compiled application, just added the permission on AndroidManifest.xml

uses-permission android:name="android.permission.ACCESS_FINE_LOCATION"

On Android on app authorization enable position.

tushe, 2017-11-27:
It would be nice to have a dialog box to enable the authorization.
It could be the solution:
https://github.com/dpa99c/cordova-diagnostic-plugin

or modify the WiFi plugin:
https://cordova.apache.org/docs/en/latest/guide/platforms/android/plugin.html#android-permissions

tushe, 2017-11-27:
It would be nice to have a dialog box to enable the authorization.
They could be the solution:
1) Add this functionality: https://github.com/dpa99c/cordova-diagnostic-plugin
2) Modify wifi plugin: https://cordova.apache.org/docs/en/latest/guide/platforms/android/plugin.html#android-permissions

closed 2017-12-11 by tushe