schollz/find #195: Autostart local FIND server at boot

alegiovannelli, 2017-11-26:
Hi everyone, I'm trying to run FIND on my sever linux Ubuntu 16.04.

If I run the command from the terminal `./findserver -mqtt 192.168.1.5:1883 -mqttadmin xxxxxxx -mqttadminpass xxxxxx: 8003` everything works regularly, if I would start the FIND server with a bash script (detailed below) at boot, I can open the FIND dashboard but the data is not updated. Some idea ?
Thanks in advance.

```
#! /bin/bash
#
cd /home/xxxxxxx/find && ./find -mqtt 192.168.1.5:1883 -mqttadmin xxxxxxxx -mqttadminpass xxxxx :8003
```

ishanjain28, 2017-11-27:
@alegiovannelli What are you using to autostart script?

alegiovannelli, 2017-11-27:
I tried with `rc.local` and now,from the Ubuntu desktop Startup sessions as from the attached picture. As I say, the dashboard is working at `http://192.168.1.5:8003` but devices data are not updated.
Thank you so much for your help.
[image]