schollz/find #196: Smartwatch support

fotiDim, 2017-11-29:
Are there any supported smartwatches? Android Wear would be a good candidate given that the needed APIs are in place.

mnlgarbe, 2017-12-07:
That should be really great.  I've been thinking on it last days.

fotiDim, 2017-12-12:
I have an Android Wear 2.0 watch now. I am willing to give a try implementing a proof of concept app given that I have some guidance on what exactly needs to be implemented. @schollz can you shed some light on this?

schollz, 2017-12-19:
@fotiDim All you need to do is find a way to gather the WiFi information (mac addresses + signal strengths). Then just format that into JSON (see https://www.internalpositioning.com/api/#post-learn) and POST it to FIND! I'm here to help if you need, but I don't have a Smartwatch so I can't play with things myself.

fotiDim, 2017-12-19:
@schollz I have 2 smartwatches that I can experiment on. I am more on iOS so far but I think I can handle this. I will give it a go over the next days.

tielur, 2018-01-19:
@fotiDim I'm interested in this as well. I have an apple watch. 

I've read that gathering the wifi data on iOS is either:

1. Not possible anymore
2. Possible with permission from Apple to use a specific API
3. Possible but you need to use a private api

Have you made any progress?

Few notes:

- I was able to find an app that Apple built that you can scan wifi ssd's and see rssi values with ( the two important pieces of data we need for FIND). So this confirms that I can still scan in the latest iOS 11.2.2 on an iPhone X.

jonathanweinberg, 2018-01-20:
Seen #108?

tielur, 2018-01-20:
> Seen #108?

Yea it doesn’t look like anything good came out of it though. According to [this article](https://developer.apple.com/library/content/qa/qa1942/_index.html) we shouldn’t be using `NEHotspotHelper` for this kind of use. Instead it recommends:

> Location — If you’d like to use Wi-Fi data to determine the device’s location, we recommend you use Core Location. This locates the device using a wide variety of techniques, including Wi-Fi. For more information, see the [Maps page](https://developer.apple.com/maps/) on the developer web site

I haven't been able to find an API in Core Location that allows us to do this though. There was mention of a private api awhile ago that people were using to get this type of information. I'm not sure of the history of that private api though, if it was completely removed or if people just don't use it because they can't publish an app with it. If it's the latter I'd be completely fine with writing my own app and deploying it to my phone without going through the app store.

tielur, 2018-01-20:
For reference that was the `Apple80211` library.

tielur, 2018-01-20:
Here's an app that allows you to scan wifi and get the information we need without a jail break. It's built by Apple though.

https://wlanbook.com/free-wifi-scanner-iphone-ipad-no-jailbreak-required/

fotiDim, 2018-01-21:
@tielur it is still on my todo list. I had minimal time to spent so far but it is on my radar for the next days.

Regarding Apple watch I believe that App Store should not be a blocker. Since this is an "enthusiast" app, I expect users to be a bit more tech savvy and be able to use Xcode to deploy directly to an Apple watch. That does not require a developer account. Essentially if you have a mac you can do this easily enough. This permits use of private APIs since the app does not go through Apple's approval process.

OmgImAlexis, 2018-04-20:
@fotiDim you don't even need xcode anymore. You can use cydia impactor to sideload even without a jailbroken device.

https://www.reddit.com/r/sideloaded/wiki/ipasigning#wiki_how_to_sideload_using_saurik.27s_.22impactor.22