schollz/find #199: can't build/compile

9H1LO, 2017-12-18:
Ubuntu 16 on x86

git clone https://github.com/schollz/find.git
cd find
go get ./...

package context: unrecognized import path "context"
Username for 'https://github.com':
Password for 'https://github.com':
# cd .; git clone https://github.com/foo/vendored /root/find/src/github.com/foo/vendored
Cloning into '/root/find/src/github.com/foo/vendored'...
remote: Repository not found.
fatal: Authentication failed for 'https://github.com/foo/vendored/'
package github.com/thinkerou/favicon
        imports golang.org/x/sync/errgroup
        imports github.com/foo/vendored: exit status 128

is there some ready x86 binaries available or better documentation or an install method (apt?) not using "GO" as have no idea of it

schollz, 2017-12-18:
Sorry! You don't need `go get ./...` anymore because all the libraries are vendored. But to use the vendored libraries you have to work in the GOPATH. I will update the documentation, but here is how you should build:

```
go get github.com/schollz/find
cd $GOPATH/src/github.com/schollz/find
go build
```

9H1LO, 2017-12-18:
wow ! quick reply ! thanks i'll try that

9H1LO, 2017-12-18:
what should $GOPATH be set to ?

root@haserver:~# cd find
root@haserver:~/find# go get github.com/schollz/find
package github.com/schollz/find: cannot download, $GOPATH not set. For more details see: go help gopath

schollz, 2017-12-18:
See https://golang.org/doc/code.html#GOPATH

Also, maybe download the latest version of Go, as it will do it automatically

9H1LO, 2017-12-19:
root@haserver:~/go/src/github.com/schollz/find# export GOPATH="/root/go"

root@haserver:~/go/src/github.com/schollz/find# cd $GOPATH/src/github.com/schollz/find

root@haserver:~/go/src/github.com/schollz/find# go build
../../boltdb/bolt/bolt_unix_solaris.go:10:2: cannot find package "golang.org/x/sys/unix" in any of:
        /usr/lib/go/src/pkg/golang.org/x/sys/unix (from $GOROOT)
        /root/go/src/golang.org/x/sys/unix (from $GOPATH)

schollz, 2017-12-19:
Did you install Go with `apt-get`? 

I'd suggest just downloading it - its easier: https://golang.org/doc/install#tarball

9H1LO, 2017-12-19:
thanks schollz !

yes as a note to all others, GO was installed via apt and it didn't work for me

i downloaded and extracted the tarball and its running fine now

managed to build and run find 

thanks again and all the best

schollz, 2017-12-19:
Great! Let me know if you have more questions.

closed 2017-12-19 by schollz