schollz/find #201: How to extract the user's location information from JSON

liNimo, 2018-01-03:
Hello FIND Team
     I have a problem for your help.How to extract the user's location information from JSON.Can you provide an  url interface for me?
     The website links as follows:
https://ml.internalpositioning.com/dashboard/indoor-nick

chucklin85, 2018-01-14:
The info on the API page would help:

https://www.internalpositioning.com/api/#get-location

So in your case, it would be:
https://ml.internalpositioning.com/location?group=indoor-nick&user=apple

Would fetch the location data for user "apple"

liNimo, 2018-01-15:
@chucklin85 Thanks for your help,I try it ,it can works.
But,I have an another question,as follow:
I know that the client and the FIND server using the MQTT protocol to transfer messages. Now there are several questions, FIND server is how to use MQTT server? How the client request data, the server is how to monitor the client? Socket used by the client and server?
Now that I'm building a WLAN network with SDN, I can extract the signal strength from the AP and upload it to the SDN controller. I want to upload the signal strength obtained by the controller to the FIND server, what should I do? In this case, I can not install APP on the phone, you can achieve the positioning of the phone.
I think this is a very interesting thing, indoor positioning does not need to install APP in the phone.

chucklin85, 2018-01-15:
Hi @liNimo, from what I can see, the client and the FIND server do not use MQTT. They use JSON based POST/GET requests over HTTP.

MQTT is there for FIND to submit events to other systems, I believe to enable the ability to do actions based on movements around the building.

liNimo, 2018-01-15:
@chucklin85 Thanks for you! it may be my mistake.
Now,FIND is using APP to upload fingerprints, and I now POST fingerprints  to FIND via SDN(Software-defined networks) controller  getting RSSI from AP instead of APP.That is, I POST fingerprints   data to FIND server , the server how to receive and resolve these data.
Can you give me some advices?

chucklin85, 2018-01-15:
I would potentially look at https://www.internalpositioning.com/plugins/

You could use this concept, which is done by the RPi in this environment, and pull apart the code to understand how they submit into the FIND Server.

You could then implement that into your SDN. 

cluster.py would be a good starting point to see how it works.

liNimo, 2018-01-15:
@chucklin85 okay,Thanks for your kindness! Later,I will try it .