schollz/find #208: New ML Algorithms

lejacome, 2018-01-24:
Hi, 
there a way to implements new machine learning algorithms(bagging, k-near neighbor, adaboost) from *.py file or I need to do it directly in golang

schollz, 2018-02-01:
The easiest thing to do is to edit the Python file. If you are still interested, I've done this a little bit already. I am working on updating FIND to include bluetooth and a whole assortment of ML algs.

lejacome, 2018-02-05:
Yes sure... I'm working implementing new algorithms and see if it's possible to mix wifi and ble to improve the accuracy system.

schollz, 2018-02-09:
Haha I also did that too! It's at GitHub.com/de0gee 

I don't have much usable documentation at the moment, but if your interested I can write something to point you in the write direction