schollz/find #210: Server crashing - panic: page already freed

sjcliffe, 2018-01-31:
I've been running find for quite a while without issues but recently it has started crashing like this:

(version  build devdevde) is up and running on http://192.168.128.128:8003
-----+-----+-----+-----+-----+-----+-----+-----+-----+-----+-----+-----+-----
DEBUG: 2018/01/31 18:15:12 mqtt.go:183: Got valid MQTT request for group stillskink, user motox
panic: page 42046 already freed

goroutine 22 [running]:
panic(0x8f8e80, 0xc8202ca040)
	/usr/lib/go-1.6/src/runtime/panic.go:481 +0x3e6
github.com/boltdb/bolt.(*freelist).free(0xc82046a210, 0x49a8b, 0x7f7e3a43e000)
	/share/lib/go/src/github.com/boltdb/bolt/freelist.go:121 +0x37b
github.com/boltdb/bolt.(*node).spill(0xc82013a2a0, 0x0, 0x0)
	/share/lib/go/src/github.com/boltdb/bolt/node.go:363 +0x2ca
github.com/boltdb/bolt.(*node).spill(0xc82013a150, 0x0, 0x0)
	/share/lib/go/src/github.com/boltdb/bolt/node.go:350 +0x12c
github.com/boltdb/bolt.(*node).spill(0xc82013a0e0, 0x0, 0x0)
	/share/lib/go/src/github.com/boltdb/bolt/node.go:350 +0x12c
github.com/boltdb/bolt.(*node).spill(0xc82013a070, 0x0, 0x0)
	/share/lib/go/src/github.com/boltdb/bolt/node.go:350 +0x12c
github.com/boltdb/bolt.(*node).spill(0xc82013a000, 0x0, 0x0)
	/share/lib/go/src/github.com/boltdb/bolt/node.go:350 +0x12c
github.com/boltdb/bolt.(*Bucket).spill(0xc820456c80, 0x0, 0x0)
	/share/lib/go/src/github.com/boltdb/bolt/bucket.go:570 +0x1cf
github.com/boltdb/bolt.(*Bucket).spill(0xc820254398, 0x0, 0x0)
	/share/lib/go/src/github.com/boltdb/bolt/bucket.go:537 +0xbc2
github.com/boltdb/bolt.(*Tx).Commit(0xc820254380, 0x0, 0x0)
	/share/lib/go/src/github.com/boltdb/bolt/tx.go:163 +0x1d4
github.com/boltdb/bolt.(*DB).Update(0xc8202d03c0, 0xc82010efc0, 0x0, 0x0)
	/share/lib/go/src/github.com/boltdb/bolt/db.go:605 +0x11d
main.putFingerprintIntoDatabase(0xc820201780, 0xa, 0xc820201791, 0x5, 0x0, 0x0, 0x150ed3b846d72503, 0xc8202cc600, 0x2, 0x2, ...)
	/share/src/find/fingerprint.go:130 +0x2b4
main.trackFingerprint(0xc820201780, 0xa, 0xc820201791, 0x5, 0xc8202ca7a0, 0x6, 0x0, 0xc8202cc600, 0x2, 0x2, ...)
	/share/src/find/fingerprint.go:227 +0xbd0
main.messageReceived(0xc8201f2000, 0x7f7e517bf5a0, 0xc8201c77c0)
	/share/src/find/mqtt.go:185 +0x2d7
github.com/schollz/org%2eeclipse%2epaho%2emqtt%2egolang.(*router).matchAndDispatch.func1(0xc82021a300, 0xc82013fd00, 0x8e4601, 0xc8201f2000)
	/share/lib/go/src/github.com/schollz/org.eclipse.paho.mqtt.golang/router.go:140 +0x2ac
created by github.com/schollz/org%2eeclipse%2epaho%2emqtt%2egolang.(*router).matchAndDispatch
	/share/lib/go/src/github.com/schollz/org.eclipse.paho.mqtt.golang/router.go:162 +0x52

schollz, 2018-02-01:
When did you download/build FIND? 

I found a bug in boltdb/bolt awhile back, and it has been fixed in new versions, but I believe this is the type of message I got back when it was broken.

sjcliffe, 2018-02-01:
I last updated & rebuilt on 12th Jan. I ended up blowing away my db file and re-learning locations as I only had around 6.

schollz, 2018-02-01:
Are you using Go1.6? Could you try to see if it resolves with the latest version of Go?

sjcliffe, 2018-02-02:
I was using go1.6.2 but installed 1.9.3 and recompiled and get the same error with my old db file:

(version 2.3 build 91ab5c29) is up and running on http://192.168.128.128:8003
-----+-----+-----+-----+-----+-----+-----+-----+-----+-----+-----+-----+-----
DEBUG: 2018/02/02 17:10:16 mqtt.go:183: Got valid MQTT request for group stillskink, user motox
panic: page 42046 already freed

goroutine 25 [running]:
github.com/boltdb/bolt.(*freelist).free(0xc42029e030, 0x49a8b, 0x7fc59243e000)
	/home/steve/go/src/github.com/boltdb/bolt/freelist.go:121 +0x301
github.com/boltdb/bolt.(*node).spill(0xc42010a230, 0xc4200e63c0, 0xe155a0)
	/home/steve/go/src/github.com/boltdb/bolt/node.go:363 +0x210

Interestingly my old db file is 165MB whereas the new one I created yesterday is only 2MB. Is it normal for them to grow this big over time?