schollz/find #211: Cannot login: group does not exist on local server?

Nanne118, 2018-02-08:
After a lot of swearing and trying things I have finally managed to succesfully install go, MQTT and find (server) on the raspberry pi 3. I can also succesfully send the command to make a group (curl -X PUT "192.168.19.103:8003/mqtt?group=groupnanne". I get the response that the group has been created, I get the password for the group and if I try to make the exact same group again it tells me your password exists and it relists the password.

However, I cannot access the group from my phone, other raspberry pi (with ARM client), from my laptop or indeed even login to the group on the browser at 192.168.19.103:8003. The phone gives an error: undefined, the windows client also spews out errors and my arm client on raspi keeps sending the fingerprint to the ml.internalposition.com  , as changing the URL apparently is not possible on ARM?

Nanne118, 2018-02-08:
I was able to get the arm client pointed towards the correct server IP:PORT, but it spews out another error:
12:37:45:727 main - crit 00a  Post 192.168.19.103:8003/track: unsupported control scheme ""

schollz, 2018-02-09:
Yeah that looks like it's still not connecting. Are all your devices on LAN or Wifi? All devices need to be on the same network to be accessible

jeffreychan121, 2018-06-04:
i also think so.