schollz/find #213: Possible to create "heading towards"?

CrispinP, 2018-02-09:
Awesome tool! I have multiple APs in the house and it works well to work out where I am.

Is there a way I can make it predict where I am heading towards?
Conceptually, it would be the one which is increasing in accuracy.

schollz, 2018-02-09:
This is a great idea. This could also be predicted by doing some machining learning on the actual routes.

However, there is a limitation because the latency in collecting a fingerprint (a few seconds) might not be fast enough to make it useful to predict where you'll be as you'll already be there.

CrispinP, 2018-02-10:
It would of course have a margin of error but for the most part, it would be OK in my case :)