schollz/find #217: Hi! this looks awesome.

rpeders, 2018-03-06:
**Im wondering if its possible to install this on the same pi as hass.io (docker)
If yes: how would i do that?**
I'm not skilled with code and stuff. But this is how i use find now:
I have room+, tasker and klwp working together.
So my homepage on my phone change depending on wich room i'm in
I can control my livingroom tv from home screen when in livingroom and a different menu 
to control my bedroom tv when i'm in my bedroom.

schollz, 2018-03-06:
Yeah, you can just use the docker image.

`docker pull schollz/find`

rpeders, 2018-03-06:
wiill that work when i have a mosquitto server running in hass.io addons?

schollz, 2018-03-08:
The docker image will run its own instance of mosquitto, so it won't interrupt your mosquitto server.

closed 2018-04-24 by schollz